satnogs/satnogs-rotator #12: worm geear design

papamat, 2015-04-25:


papamat, 2015-04-25:
calculate tooth width

zisi, 2015-06-14:
* Gear ratio: i12 = 30
* Angle between axis of gears: δ = 90 deg
* Number of threads in worm: If i12 >= 30 then z1 = 1
* Number of teeth in worm wheel: z2 = i12*z1 = 30
* Center distance: initial case a = 45.5 mm
* Worm reference diameter: AGMA d01>= 11.5*(a/25.4)^0.875 = 19.15 mm, so d01 = 19.5mm
* Worm wheel reference: d02 = 2*a - d01 = 71.5 mm
* Axial module: ms = d02/z2 = 2.38 , so ms = 2.5
Recalculate d02, a with new axial module
* d02 = z2*ms = 75mm, a = (d02+d01)/2 = 47.25mm
* Axial pitch: ts = π*ms = 7.854mm
* Reference lead angle: γ0 = atan(d02/(i12*d01)) = 7.3 deg
* Worm tip diameter: dk1 = d01 + 2*hk = 24.5mm
 * Worm teeth reference addendum in axial section: hk = hk* *ms = 2.5mm
  * Worm tooth reference addendum coefficient: hk* = 1
* Worm root diameter: df1 = d01 - 2*hf = 13.5mm
 * Worm tooth reference dedendum: hf = hf* *ms = 1.2*ms = 3mm
  * Dedendum coefficient: hf* = 1.2
* Worm length: L = 2.5*ms*sqrt(z2+2) = 35.36mm
* Worm tooth thickness: smx1 = smx1* * ts = 3.927mm
 * Tooth thickness coefficient: smx1* = 0.5
* Normal pressure angle: aon = 20 deg
* Worm wheel throat diameter: dk2 = d02+2*hk = 80mm
* Worm wheel root diameter: df2 = d02 - 2*hf = 69mm
* Worm wheel outside diameter: de2 = dk2 + 2*mx = 83.5mm
 * Worm wheel tooth external addendum: mx = n*ms, 0.4<=n<=1.5
* Effective worm wheel face width: b2H,max = sqrt((2*a - df2)^2 - (2*a - de2)^2) = 23mm

closed 2015-08-04 by ppapadeas