satnogs/satnogs-rotator #19: Assembly.fcstd crashes FreeCAD

cshields, 2015-05-31:
The src/Assembly.fcstd file doesn't load in FreeCAD, in fact it crashes it hard.  Reproduced in v0.14, in OSX and Linux - then upgraded to v0.15 on both and reproduced again.

This is likely a FreeCAD problem, but noting here as I can't seem to get this file to work.

(lots of repeat, only pasting the end)
<unknown exception traceback><type 'exceptions.ImportError'>: No module named importPart
<unknown exception traceback><type 'exceptions.ImportError'>: No module named assembly2lib
<unknown exception traceback><type 'exceptions.ImportError'>: No module named assembly2lib
<unknown exception traceback><type 'exceptions.ImportError'>: No module named assembly2lib
*** Abort *** an exception was raised, but no catch was found.
	... The exception is:SIGSEGV 'segmentation violation' detected. Address 21

cshields, 2015-05-31:
Should have poked google a bit before filing - I see the following mod is needed for this file,
https://github.com/hamish2014/FreeCAD_assembly2

However..  it still crashes with the same SIGSEGV..  Is there anything else I am missing?

zisi, 2015-07-22:
Hey, we made some changes in assembly file, could you please try to opened again?

cshields, 2015-07-23:
Tested/verified in FreeCAD .15 OSX - no crashing anymore!

thanks!

closed 2015-07-23 by cshields