satnogs/satnogs-rotator #21: Research a better PoE supply for the rotator

acinonyx, 2015-06-09:
proposals:
802.3af compliant or compatible
48V

zisi, 2017-01-09:
Finally, the ground station needs 48V@1A.

bittorf, 2017-01-10:
Using standards is generally a good idea 8-) you can use Poe+ 802.3at
which can deliver >50 watts. The upcoming 802.3bt allows even more. Maybe that is an option?

zisi, 2017-01-21:
Thanks for the comment!!
As i know 
802.3at (the update for 802.3af) is up to 25.5W and using 2 pairs.
802.3bt is up to 55W and using 4 pairs (it requires PoE splitter).  
A good choice is ubiquiti POE-50-60W.

bittorf, 2017-01-22:
@zisi: the POE-50-60W looks interesting (cheap and powerful enough), although i think it's not really 802.3bt, is'nt it? the datasheet just says Pins 1, 2, 4, 5 (+) and Pins 3, 6, 7, 8 (-). so the question is more: can we built a compatible receiver?

zisi, 2017-01-25:
@bittorf: maybe it's not standard PoE, i can't find more information about it. I found this https://cds.linear.com/docs/en/datasheet/4295f.pdf in order to build our PoE PD, but i want to read and search more.