satnogs/satnogs-rotator #22: Initial v3 Imperial review

ppapadeas, 2015-06-16:
Requesting review for https://github.com/satnogs/satnogs-rotator/commit/7370bba58792ccf4114aa23b8c2352eda416b61e and latest BOM.

etihwnad, 2015-06-16:
.. latest BOM is now in decab30f8e079bd182559642e782c4d1aa37d8dc
looks good so far!

cshields, 2015-06-17:
You mention imperial review - I took a quick look and the sizes of the src files for this commit are all still for metric.

For imperial PVC we need a 17mm hole.  I have the parts resized for this in https://github.com/cshields/satnogs-hardware/tree/imperial - don't think that any of the 4 relevant parts changed here..  hmm maybe the homing ring but that is an easy adjustment.

Also, the Assembly.fcstd still crashes freecad for me, even after the latest change unfortunately  (issue #19 )

One minor issue on the BOM, the T slot L102 (row 15) total does not seem to match the qty * price?

Excited to see this nearing completion!!

etihwnad, 2015-06-18:
Having the screw holes and nut pockets compatible with both metric and imperial sizes would be a good feature.  

The screw diameters are fine to use the metric sizing since they are all larger:
M3 -- #4 (2.9mm dia)
M4 -- #6 (3.5mm)
M5 and M6 -- #10 (4.9mm)

It is the nut pockets which should be increased a bit to fit the imperial hardware.
Nut dimensions across the flats (wrench size):
M3 = 5.5mm  vs.  #4 = **6.4mm**
M4 = 7.0mm vs. #6 = **7.9mm**
M5/6 = 8mm/**10mm** vs. #10 = 9.5mm

from:
http://www.fairburyfastener.com/xdims_metric_nuts.htm
http://www.fairburyfastener.com/xdims_nuts_mch_scr.htm

Proposal
-------------
* Limit the use of M3 hardware for compatibility and prefer M4 or larger.
* Size holes and pockets to accept either metric or imperial hardware with no modifications

Though US hardware stores carry a few metric items, they are light on variety.  All 3 of the ones near me (Menards, Home Depot, and ACE Hardware) only have M3 screws in 6, 10, and 20mm length and no M3 threaded rod.  Therefore, screwing the M3-threaded stepper motor to a mounting plate is no problem with v3, but not with the threaded rod.  They also don't carry #4 threaded rod, but do have #6 and up.  ACE did have M4 and M6 threaded rod.

Also, most stores are light on #4 size parts.  There are plenty of options in #6 and larger sizes.  Sure, it's possible to order all the hardware bits online, but it greatly increases the effort to put the parts together unnecessarily.

With this change, and having compatible hole/pocket sizes, only the bore holes for tubing would then need to change for the imperial version.

cshields, 2015-08-01:
I would recommend the option of SAE hardware be pushed back further on the roadmap..  The metrics parts *are* available in the US, though like Dan said they are online.  My counter-reasoning for lowering the priority of this is that there are other parts in v3 that are even less likely to be found locally (like the 2020 t-railing and associated parts, bearings, etc)..  Might be better for now to come up with some US-centric BOM links that make it easier to find the parts.   I scoped out all of the metric hardware on www.boltdepot.com for $22 including shipping.  (this was minus the 2020 slot 6 parts, that's another story, I kicked off a thread in discourse about)

Plus, this will mean fewer 'customized' parts for imperial measurements.  To handle the PVC difference in v3 we are down to just 4 parts that differ from the metric version.

cshields, 2015-08-02:
..  I should also add that in terms of an "imperial review" of v3 I am working on building my own here using the same specs except for the 4 files I've modified for imperial pvc..  So far they are working but I won't be done for a couple of weeks.  If they work we can release these files along with the rest of v3.

ppapadeas, 2016-10-18:
Any updates based on the latest pushes on this?

ppapadeas, 2017-01-30:
After thorough consideration we will not be maintaining a separate imperial version for satnogs rotator. In case local sourcing for rotator parts are not possible minor adjustments might be needed, which could be tracked in a separate branch. (feel free to contribute)

closed 2017-01-30 by ppapadeas