satnogs/satnogs-rotator #45: imperial stl files missing from v3 release

cshields, 2016-02-23:
Although we advertise v3 as having the imperial files [here](https://github.com/satnogs/satnogs-rotator/releases) the files are not in the package after all.  In addition they seem to be missing from the repo .

The relevant 4 files can be found at https://github.com/cshields/satnogs-hardware/tree/imperial-merge and were originally merged in with https://github.com/satnogs/satnogs-rotator/pull/29

zisi, 2016-02-23:
Hey, we updated the release v3 for 32mm axis. Thanks! :rocket: 

Can you check it?

cshields, 2016-02-23:
yeah, I'm totally lost now.  We need the ID to be 17mm radius and I see (looking at the collar) that it is now 20.4?

zisi, 2016-02-23:
Here https://github.com/satnogs/satnogs-rotator/releases, in v3.0-32mm.zip we have added stl files for imperial parts. In the src files we haven't added the src for imperial because is the same with metric with difference in diameter. In master branch are placed the working version of rotator that now it has 40mm diameter axis. :rocket:

cshields, 2016-02-23:
What is the inner diameter used for the 32mm parts?

zisi, 2016-02-23:
is 32.4mm

closed 2016-02-23 by papamat