satnogs/satnogs-rotator #46: Investigate gears slipping

zisi, 2016-03-23:


etihwnad, 2016-03-23:
What gear is slipping?  The worm gear and shaft connecting the pulley?

If so, a few students are working on a solution.  Demo by Hamvention for sure, and pics before.  It will also reduce the assembly steps.

zisi, 2016-03-24:
When the (heavy) antennas are back mounted, the axis gear is slipping in the mast, as a result the axis gear breaks in the nut pocket area.
One solution is to open holes (2mm) in the mast, in the same points with the M3 set screws of axis gear and increase  the number of set screws (mounting points).
When the nuts of worm axis are not screwed tightly, the upper nut is unscrewed after extensive run.
One solution is to use allen wrench instead of threaded rod.

This is your solution, https://huboard.com/satnogs/satnogs-rotator#/issues/117456694 ?

etihwnad, 2016-03-27:
> When the nuts of worm axis are not screwed tightly, the upper nut is unscrewed after extensive run.
> One solution is to use allen wrench instead of threaded rod.
> 
> This is your solution, #34?

Yes, #34 is the [proposed] solution.  To keep the pulley tooth profile easy to print, maybe move to a XL timing belt which has a 5.08mm pitch.  Or GT2 3mm pitch.  Or...

> One solution is to open holes (2mm) in the mast, in the same points with the M3 set screws of axis gear and increase the number of set screws (mounting points).

How about shaft collar type attachment for the axis gear to the mast  like this:  http://www.onlinebearings.co.uk/images/P/DSSC300.jpg

Half of the clamp is printed with the axis gear and the other half is printed as a separate part.  M5 bolts to clamp the halves together around the mast.  Something like 10mm to 15mm thick.  As a secondary slip protection, have a long bolt go through both halves and the mast 90deg to the split direction to pin the connection.

zisi, 2017-01-02:
Closing in favor of issue #34, issue #49, issue #50, issue #51.

closed 2017-01-09 by ppapadeas